NARUTOfzr/Neptune_3 #1: Filament run-out pin?

mlee12382, 2022-08-02:
I don't see the pin assignments for the run-out sensors? I also noticed that you have the drivers defined as a4988 instead of tmc2208_standalone.

NARUTOfzr, 2022-08-04:
[image]
This is the source file provided by makerbase. From the PCB schematic diagram, the run-out pin is PB4.
The drivers defined as a4988 does not seem to affect normal use, because UART mode is not used.

mlee12382, 2022-08-04:
Awesome, thank you. Wouldn't they still be able to utilize stealthchop even without uart? I also ran into an issue with trying to define the screen to the alternate blue theme, seems there may be some missing files or definitions somewhere to make it work? Any thoughts on that? I am also trying to work on a configuration for the Neptune X but running into issues with the flash not completing, I'm sure I'm missing some critical board settings somewhere. Do you have working configs for the X?

NARUTOfzr, 2022-08-04:
No, there aren't UART cannot utilize stealthchop. 
For neptune3: Defining the screen should require the source code of the screen, but unfortunately I don't have it either.  At present, I don't know whether I will open source the screen source code.  If it is open source in the future, I will put it on my GitHub for everyone to use. 

At present, I haven't tested marlin2.0 on neptune_x motherboard.  The specific configuration method is not clear.  I will study it in depth when I have time.

NARUTOfzr, 2022-08-04:
> Check through my fork of your code, it's not working for the other printers for some reason but the N3 seems to be working, I've got a couple of users testing it and results seem promising. Also note the Neptune 2 configs other than lvgl do work separately, I was trying to combine everything in one place.

This is about the marlin2.0 configuration method of neptune2 series motherboards. It may help you.
[Neptune 2 series marlin configuration.docx](https://github.com/NARUTOfzr/Neptune_3/files/9261134/Neptune.2.series.marlin.configuration.docx)

mlee12382, 2022-08-04:
I have working Neptune 2 /2S /2D setups, I tried to merge them with your Neptune 3 source code and they don't compile, I've got them set up with selectable definitions so it's an easy matter of selecting a specific setup and hitting build instead of having separate config files for each printer. I will play around with a fresh set of marlin source and porting your code and see if that helps. They work well separately I'm just trying to make it more convenient for sharing lol. For the X I was told by another elegoo employee that it's a modified Robin nano v3 and I have the board layout which has some of the pins defined that I was trying to build from, it was missing pin-outs for the screen port, which may have been part of my issue.

Another thing I'm wondering is if the N3 with the dgus screen will be able to utilize the wifi add-on? I know the mks wifi by default only works with lvgl interface. Is there a workaround?

NARUTOfzr, 2022-08-06:
About source code merging. Since I am not the author of the source code nor the software developer, I am not sure whether it can be merged. I believe you will be more professional than me!

Just now, I updated the source code. The previous source code enables the test mode, and the compiled firmware will have some strange problems (bugs). Moreover, the automatic leveling device is unstable due to the feedback of more users. So I fixed some bugs that have been found and modified the leveling method and speed (now the user feedback that the automatic leveling is more stable than before). I hope you can update the source code and recompile the new firmware for use.

In principle, the motherboard of Neptune 3 can also use the WiFi module. However, since the screen does not define this part, the WiFi experience will be less friendly. My idea is this: you can use the AP mode of the WiFi module to configure the connection to the router using the sta mode; Then find the IP address of the printer WiFi in the router. Thereby realizing the function of using WiFi.

About Neptune x-marlin configuration. Defining the screen portion can be cumbersome. I think you can refer to the motherboard of nano v1.2 (Neptune 2) to configure the screen. Because they use the same screen.

NARUTOfzr, 2022-08-06:
[image]
[image]

Neptune x. I wonder if it is the pin of this part?

mlee12382, 2022-08-06:
Yes, I was missing those for the screen. Looks like the pins are all the same as from the znp Robin nano v1.2 and v1.3, hopefully that will get me closer to making it work :) Thank you.
Do you know how to properly configure the run-out sensor on the Neptune 3? I tried setting it up with the runout_distance and motion_sensor and it only extrudes the amount that runout_distance is set to as if it doesn't detect the filament.

NARUTOfzr, 2022-08-06:
Neptune 3 has already configured the runout sensor by default. Software developers seem to use a special. H file to define it, so there is no need to repeat the definition in configuration. H.

At present. It is defined after Z height > 0.8mm. The runout sensor will work.

mlee12382, 2022-09-22:
@NARUTOfzr is there any chance that the source files for the screen firmware are available? I would like to look at them in the DGUS Tool so I can learn how to make my own screen firmware.

NARUTOfzr, 2022-09-24:
> @NARUTOfzr is there any chance that the source files for the screen firmware are available? I would like to look at them in the DGUS Tool so I can learn how to make my own screen firmware.
@mlee12382 Hi！Michael.This is the screen source file. Hope you can use it as a learning tool because we haven't made it open source yet. If you want to modify the contents, please do not change/delete the original layout and parameters (can add operation button, but can not delete the original button, because considering the compatibility between the screen and the motherboard). Also, please do not advertise and share without verifying the actual usability, as this may cause trouble for our after-sales team.
And thank you for your support of ELEGOO products, thank you!

Note: I will delete the file after you get it (or 24 hours later).
[2_ZNP_DGUS_Project.zip](https://github.com/NARUTOfzr/Neptune_3/files/9637603/2_ZNP_DGUS_Project.zip)

mlee12382, 2022-09-24:
> 

Thank you very much for the files :) I will be cautious with sharing any files before proper testing. I very much appreciate the help! :) Which version of the DGUS tool are you using for editing the files?

NARUTOfzr, 2022-09-24:
@mlee12382 
Here may be the information you need:
http://www.ampdisplay.com/download1.php?cat=HMI%20UART(DWIN)&sub_cat=DGUS

djjuno, 2022-10-06:
Hello @NARUTOfzr Can you please provide us with the Kernel for the Elegoo screen as well?
Thank you!

NARUTOfzr, 2022-10-08:
@djjuno 
Aha! Did you also update the wrong boot loader?

djjuno, 2022-10-08:
> @djjuno Aha! Did you also update the wrong boot loader?

Yes, It's very possible. I just need everything that will put the screen back to stock. I purchased a PGT05 and a HDL662 to repair the screen. So, I just need the original OS and UI files that are used to load the screen so I can flash the correct ones that work with the project.

djjuno, 2022-10-08:
@NARUTOfzr Going off of the manufactures website. There are 2 OS kernels for the screen. DGUS2 and the TA kernel. Do you know which kernel Elegoo is using?

NARUTOfzr, 2022-10-09:
@djjuno 
All right! I don't know how you operate to cause this problem. If you can provide some details, I think it will be better to help you solve this problem.

If you just updated the boot loader file on another site so that it can't be touched or the screen is blank, just place the "T5L_UI_DGUS2_V36_20201127.BIN" file in the "DWIN_SET" folder. Then update the firmware according to the Update Firmware instructions.

Tip: The cable connecting the screen to the motherboard. If the TX/RX cable is not in good contact, the screen cannot receive the startup signal from the motherboard, which will also cause the screen to fail to start normally.

[V36内核1.zip](https://github.com/NARUTOfzr/Neptune_3/files/9740499/V36.1.zip)

djjuno, 2022-10-20:
@NARUTOfzr Thank you for the information. This is going to be a LONG and MESSY post, so I apologize in advance :)

I have had some bad luck with the screens. My 1st printer screen stopped working I think the screen just died. I purchased 2 printers originally and tested with the working screen and it came up just fine. Elegoo was kind enough to send me a new screen replacement and that's working fine.

I'm probably making myself sound like a HUGE idiot but I'm really not. Printer firmware and Screen UI / firmware are a new subject for me, and I am just trying to learn. When I have asked Elegoo and yourself for help. I think I'm not explaining things correctly because I'm confused of how things work and because of that. I'm making everyone else confused. I'm going to try and explain so hopefully it makes sense. I have spent MANY hours trying to figure this out and right about now I'm feeling pretty stupid. So here it goes.

I wanted to learn how to create my own project and help with functions and firmware and I think I messed up my second screen pretty bad. I did some research, and I ordered a PGT05. After watching some videos, I tried to get the screen to come back to life. using the PGT05. I was able to get the screen to turn on and there were a bunch of lines in it. I was able to load a demo I found. It loaded but the lines were still there. To be honest. I have no clue how I got the demo to load but I did. The touchscreen also didn't work anymore when it booted up.

Here are the examples:
[image]
[image]

Before I asked for help. I tried to flash the OS kernel from the DWIN site. [Kernel Update](https://www.dwin-global.com/kernel-upgrade/) "T5L_OS_V20_RTC(RX8130)" and then the UI "T5L_UI_DGUS2_V53_20220917" kernel but I think I may have flashed the wrong kernel and messed it up beyond repair " I'm trying not to give up though".  Anyway, I watched this video from the DWIN site " [PGT05 Burning Video](https://forums.dwinglobal.com/index.php/forums/topic/%e3%80%90share%e3%80%91t5l-burner-operation-video/) But I am confused because I thought she was burning the OS Kernel and then a UI Kernel using the PGT05. After that you use the SDCARD to load the project files and update the screen as normal with the original project firmware. This is why I originally asked you for an OS kernel for T50 chip. There are so many different versions of instructions for the PGT05 burner and different names they say to rename it to. It's really easy to mess up things. If you could help me out. That would be fantastic! :)

Here are the questions I have. Some of them may sound like the same thing. Please correct me if I'm wrong.

1. What kernels are on the screen when it comes from the factory?
- I plugged in the working screen into the HDL662 and I was able to read the following:
[00:09:31.628]->R: 5a a5 06 83 00 0f 01 36 10 Going off of the development guide. That's saying it's UI v36 and OS v10 If I'm correct.
2. Do you know what files I need to flash/burn to get the screen back to its original state with the PGT05?
3. Do I need to burn an OS kernel as well as a UI kernel or does it just need the UI kernel?
4. Do you know, or can you find out what file names I need to rename for the PGT05 erase the chip and then flash the correct kernel. 

Here are the steps that I have found and questions I have about the process.

Going off of the steps of this video [PGT05 Burning Video]
(https://forums.dwinglobal.com/index.php/forums/topic/%e3%80%90share%e3%80%91t5l-burner-operation-video/)
Here are the instructions provided: 
[Instruction manual.docx](https://github.com/NARUTOfzr/Neptune_3/files/9826036/Instruction.manual.docx)

Step 1:
- rename the correct kernel to "T5UIC3_Flash and T5L.BIN"
- Burn the PGT05 with the Erase kernel. Then burn the kernel to the screen.
My Question: Is this the OS kernel or the UI kernel? Or is it a special kernel file to erase the screen to allow the UI kernel file to be loaded.

Step 2:
- Rename the UI kernel to T5UIC3V10.BIN
- Burn the PGT05 with the UI kernel and the burn the kernel to the screen

Step 3:
- Copy the project files to the DWIN_SET folder on the SDCARD. Insert the card into the screen. Turn on the screen and it should update the screen with the project
My Question: Do I need to keep the "T5L_UI_DGUS2_V36_20201127.BIN" file in with the project files in the DWIN_SET folder or do I take it out and just update without the bin file?

[image]

OR

[image]

I just think I'm missing something, or not burning the correct files or I have fried the chip at this point. 

I'm going to order 2 or 3 "ZNPCO43_01WTR" screens from Elegoo so I can replace them. But I would really love to learn how to do this so I can develop updates to the UI.

I have also ordered 3 DMG48270C043_04WTR Screens which I wanted to try to load the Elegoo Custom Software on. I wasn't sure if it was going to work but this is the model of screen that's in the Elegoo firmware document. I figured I would give it a shot. I ordered 3 of them because they are coming from China, they are pretty inexpensive, and I have messed this other one up and if I mess these new ones up. I have extra to mess around with.

Thank you for even trying to read this!

djjuno, 2022-10-20:
@NARUTOfzr By some act of god. I was able to get the firmware on the screen. The only issue I have now is that the touch is not working.  [Screen Video](https://drive.google.com/file/d/15EuOzh8jpBkozJC8pXVDYDTRZvLZpsql/view?usp=sharing) I have tried to just flash the file that you gave me. It's still not enabling touch. I know the touch screen is working. I was able to do a touch configure in another demo. But it's just not working when I load the Elegoo Project files. Is there a config that will erase everything so that it makes sure that it will flash everything new?

NARUTOfzr, 2022-10-20:
> @NARUTOfzr By some act of god. I was able to get the firmware on the screen. The only issue I have now is that the touch is not working. [Screen [...]

@djjuno Based on what you've done so far, it's usually not going to damage the screen (unless you short-circuit somewhere).
After watching your video, I think it may be because the file T5L_UI_DGUS2_V36_20201127.BIN is not loaded in the file, so it cannot be touched normally. You can take a picture of the interface displayed after loading the firmware and feedback it for confirmation. If you want to learn the development process of DWIN screen, maybe you can directly consult DWIN's after-sales staff for more relevant information. To be honest, I am still learning. I'll share what I know with you. As for the question you asked, I will also follow up with the manufacturer DWIN and tell you after confirmation.

djjuno, 2022-10-20:
@NARUTOfzr Yeah, I haven't had any sparks yet :) Just maybe a bruised forehead from hitting it on the desk multiple times! I think the PGT05 helped with getting the screen to come back to life, but it didn't end up solving the issue. 

The steps that I can remember doing "It was really late" was loading the v10 OS on SDCARD and then directly flashing to the screen. Then I uploaded a factory demo project for the DMG48270C043_04WTR to the screen using the v10 OS and it loaded it up no screen display issues but the touch still didn't work. I then loaded a config to do a touch calibration. The touch calibration worked on the update screen. It just didn't work when I loaded into the demo or the custom elegoo project. If I load the screen up and use the HDL662 and debug the screen in DGUS. I can see that it registers the touch location and will push out different readings based on where I touch the screen and run the vp command to see the updated location. So, I'm 100 sure the resistive touch layer is working. I just think the link is broken between UI and firmware somehow.

When I did a firmware update. I did notice the numbers changing based on when I was loading a new config or UI. So, I'm pretty sure the config is updating. I'm just not sure it's fully clearing out the config or I'm still doing something wrong.  I'm a lot further than I have been in a week so I'm just extremely happy to know the screen wasn't damaged to the point of no return and all the time that I have spent wasn't a complete waste. Defiantly given me a better understanding on how the device functions.

If you could reach out to DWIN to find out the if there is a wipe config. As well as finding out the correct combination of OS and UI, I should be using for the ZNPCO43_01WTR that would be fantastic! I'm not going to receive my DMG48270C043_04WTR screens for another week or so. So, I can't test it out on those yet. I will get those pictures to you soon with the different stages of the update screen.

NARUTOfzr, 2022-10-21:
https://drive.google.com/file/d/15EuOzh8jpBkozJC8pXVDYDTRZvLZpsql/view

NARUTOfzr, 2022-10-21:
https://drive.google.com/file/d/15EuOzh8jpBkozJC8pXVDYDTRZvLZpsql/view
@djjuno  Can you now restore the screen to this state in the video? Format TF card if possible (must be :4096bit/FAT32)
[image]

Then unzip the "底层文件.zip", put the "DWIN_SET" in the TF card root directory, insert it into the TF card slot of the screen, power on and download the firmware (as shown in the figure, PGT05 is not needed). Take out the TF card, connect the printer, restart the machine and try to see if there is a touch response.
[image]

[底层文件.zip](https://github.com/NARUTOfzr/Neptune_3/files/9839223/default.zip)

djjuno, 2022-10-22:
@NARUTOfzr I updated the files. I did see the same screen. There was still no touch after the files were updated.
[image]
[After Files Updated](https://drive.google.com/file/d/15QUpot3pkIcVNDTIAWJmoosNoTXGSQPv/view?usp=sharing)

NARUTOfzr, 2022-10-22:
@djjuno 
Try this profile with touch calibration. In the same way, place the CFG file in the "DWIN_SET" folder (delete other files).
[T5LCFG_C1_竖屏_开启校准.zip](https://github.com/NARUTOfzr/Neptune_3/files/9843594/T5LCFG_C1_._.zip)
Screen calibration Screen steps: Before displaying "End!" Then click ①②③④ in the four corners of the screen (as shown in the picture).
[image]

If you still can't touch it, try the following method to erase the kernel and then burn it again. Finally, the TF card is used to load the screen firmware of Neptune3.
[Kernel & erase files.zip](https://github.com/NARUTOfzr/Neptune_3/files/9843635/Kernel.erase.files.zip)

These are some methods provided by DWIN manufacturer I consulted yesterday. I also asked them to provide me with the PGT05 burner and the Chinese description document. I think I still need to make some attempts to make sure that the method provided is effective.
[PGT05烧录使用说明.docx](https://github.com/NARUTOfzr/Neptune_3/files/9843636/PGT05.docx)

Have a great weekend!

djjuno, 2022-10-23:
@NARUTOfzr Thank you for this! I got it working again! I loaded up the first config and it didn't give me a chance to do the touch configuration before it reached the end. You might have to do it after the config was loaded but I turned it off before then and tried the PGT05 Method. I used the same method as the instructions, but touch wasn't restored after loading. I loaded the first config again. Waiting until it was done loading. Then I touched the screen in the 4 corners. Once I did that. I removed the SDCard. Booting the screen back up while connected the printer and the touch function was restored!

[Touch Restored!](https://drive.google.com/file/d/15Ylxl1-uTyzN5mhUI3A3XwjEknKO7fmt/view?usp=sharing)

Can't believe that it's working again. I have tried so many things to get it back up and running. I pretty much lost hope at this point. Thank you so much for your help on this. When I have some time. I will try and make a video showing the steps to restore the screen back to original function. 

I hope you have a great weekend as well! Thank you again for all of your help! Now I have some bigger DWIN screens I want to see if I can get working on the project. I at least know how to get myself out of a hole if I mess something up. I need to try and figure out how DGUS works more and see what I can do with. I have to say that I'm impressed with the DWIN hardware even though I gave it a beating and it still came back. As far as the WIFI issue. Is that an issue with the printer board firmware itself? Or is it an issue with the DWIN OS and UI version? Is it possible to use the newer version of the OS and UI to get the Wi-Fi working if it's setup in the project?

NARUTOfzr, 2022-10-24:
Cheers!  That's a good start.
Question about WIFI module.  Because the machine did not advertise WiFi functionality, this was not considered when making the software.  However, according to the circuit design layout of the motherboard, the motherboard supports the expansion of WiFi module.  However, since the firmware and UI version of the motherboard do not configure the WiFi option, the WiFi function cannot be used normally.  If you update the UI version with the option to configure WiFi and properly configure the motherboard firmware, you should be able to get WiFi to work.

I don't really recommend the WiFi module, because the file transfer speed is too slow, and there may be strange bugs.  I don't know if you know you can control the printer with your Android phone? All you need is the octoprint APP and an OTG.  If you have a phone to spare, this is a nice way to wirelessly control it.

djjuno, 2022-10-24:
Yeah, with the WIFI. I really just want to get it working just to get it working. Kind of a personal project. It may not be all that useful. I just want to see if I can do it. This is my first Elegoo FDM printer, so I never used the Wi-Fi on any of the other previous versions. I have already hooked up the printer to octprint using a raspberry pi and everything works well except the usb file transfer to the sdcard.  I didn't even know that octo4a even existed! That's pretty awesome. I do have a bunch of old phones I can mess around with and see how it works. For people that don't have a bunch of pi's laying around. That's a pretty great solution for using older hardware and giving people a way to wirelessly control their device.

I'm glad I was able to finally get a better understanding of the whole os/ui/config thing. It finally clicked so I understand the relationship a little better now. When you are going though development guide after development guide and readme after readme. It really doesn't make it clear for someone who hasn't messed around with the screens before and starts to all mesh together. I didn't understand what I was reading until you gave me that last bit of information. I don't know if I'll ever write my own OS for the screen :). But at least I get what it's doing. I'm going to mess around with DGUS a little more. I might need to see if I can find an old Neptune project for an older printer with Wi-Fi and see how that's setup to configure it correctly in the project. But it would also be nice to add functions like PID Turning, etc.

mlee12382, 2022-11-11:
@NARUTOfzr is there any chance you can get the Neptune X firmware source code?

mlee12382, 2023-01-15:
@NARUTOfzr are the CAD models for the Neptune 3 available? I saw the Pro source materials were posted on Elegooofficial github. Any chance the regular Neptune 3 files are available? 
Also I would still like the firmware source files for the Neptune X if possible.

mlee12382, 2023-01-16:
Awesome thanks 😊

bmorcelli, 2023-01-16:
@NARUTOfzr are the source files of the screen firmware of Neptune 3 available? Can you share with us?