NARUTOfzr/Neptune_3 #2: Enable support for MKS Wifi module

wolllis, 2022-09-05:
Hello @NARUTOfzr,

It would be nice to have this feature active like in the Neptune 2 printer :)

I noticed the support for MKS Wifi module being disabled in the project configuration.
Is there any particular software/hardware related reason for this? If its just not tested then I may support here.

Best regards,
wolllis

NARUTOfzr, 2022-09-10:
The hardware of this motherboard supports the extension of MKs_wifi module. However, current firmware does not support this feature.

NARUTOfzr, 2022-09-10:
> Hello @NARUTOfzr,
>
> It would be nice to have this feature active like in the Neptune 2 printer :)
>
> I noticed the support for MKS Wifi module being disabled in the project configuration. Is there any [...]

The hardware of this motherboard supports the extension of MKs_wifi module. However, current firmware does not support this feature.

wolllis, 2022-09-17:
Thanks for clarification. Hope it will be added somewhere in the future,
Cheers

closed 2022-09-17 by wolllis