NARUTOfzr/Neptune_3 #3: Printer resets at print start

SebSL83, 2022-09-08:
Hi,

i updated to Firmware V1_1.0.3 an UI Version V1
and now when i start a printing, the Nozzel and the bed heating up. 
than the Printer starts Homeing, after he move in the center and back to left front coner to start the
print, he stand still for 1 sec and than the display goes out, the printer reboot and nothing happend.

I tried diffrent Printcodes, all the same issues.

SebSL83, 2022-09-08:
Hi,
the CubeElemantal is the gcode i would print. the Buddha is the original Eleggo that came with the printer.
it also makes the same issues. i try your ocarina gcode and give feedback.

[CubeElemental.zip](https://github.com/NARUTOfzr/Neptune_3/files/9528693/CubeElemental.zip)
[EN3_The Buddha.zip](https://github.com/NARUTOfzr/Neptune_3/files/9528694/EN3_The.Buddha.zip)

SebSL83, 2022-09-08:
with the ocarina gcode the same happens. i have done a short clip you can see after homeing and move to start printing the
printer restart himself.

[short_Video.zip](https://github.com/NARUTOfzr/Neptune_3/files/9528920/short_Video.zip)

SebSL83, 2022-09-08:
> > with the ocarina gcode the same happens. i have done a short clip you can see after homeing and move to start printing the
> > printer restart himself.
> > [short_Video.zip](https://github.com/NARUTOfzr/Neptune_3/files/9528920/short_Video.zip)
> 
> Did you try reflashing the firmware?

i have reflashed the borad firmware for 10 min. issues still there. the display firmware will i reflash now.

mlee12382, 2022-09-08:
> i have reflashed the borad firmware for 10 min. issues still there. the display firmware will i reflash now.

Idk if it will fix your problem but try my 1.0.4 build https://github.com/mlee12382/Elegoo-Neptune-Marlin/tree/bugfix-2.1.x/config/Neptune%203/Advanced%20Features/No%20Nozzle%20Preheat

Manpoozle, 2022-09-08:
Sorry to butt in, but Michael which screen firmware goes with the 1.0.4 motherboard firmware? Thanks

mlee12382, 2022-09-08:
> Sorry to butt in, but Michael which screen firmware goes with the 1.0.4 motherboard firmware? Thanks
> 
> > 
> > 
> > with the ocarina gcode the same happens. i have done a short clip you can see after homeing and move to start printing the
> > printer restart himself.
> > short_Video.zip
> > 
> > Did you try reflashing the firmware?
> > 
> > i have reflashed the borad firmware for 10 min. issues still there. the display firmware will i reflash now.
> > 
> > Idk if it will fix your problem but try my 1.0.4 build https://github.com/mlee12382/Elegoo-Neptune-Marlin/tree/bugfix-2.1.x/config/Neptune%203/Advanced%20Features/No%20Nozzle%20Preheat

AFAIK it's the same v1 screen firmware. I just re-compiled based on the new configs on this repo.

SebSL83, 2022-09-08:
> Idk if it will fix your problem but try my

the reflashing of the Screen dosent work. the issues is still the same. i am wondering that the auto level data still be there after reflahing board and screen...
now i will try your 1.0.4 did you change something?

mlee12382, 2022-09-08:
> > Idk if it will fix your problem but try my
> 
> the reflashing of the Screen dosent work. the issues is still the same. i am wondering that the auto level data still be there after reflahing board and screen...
> now i will try your 1.0.4 did you change something?

It's built on the new code from this repo that hadn't been compiled yet, my version does enable some additional features also though.

SebSL83, 2022-09-08:
> It's built on the new code from this repo that hadn't been compiled yet, my version does enable some additional features also though.

so after flashin your version thr printer runs :) thank you
could you explain some of your features? at the first view i see nothing new ^^
THX

mlee12382, 2022-09-08:
> so after flashing your version thr printer runs :) thank you
> could you explain some of your features? at the first view i see nothing new ^^
> THX

Unfortunately with the screen firmware they don't actually show up on the screen, but I've enabled better communication with usb connections for octoprint, s-curve acceleration is enabled, m600 for color changes is enabled but a work in progress, seems to be an issue with getting the printer to resume after the filament change, linear advance is available, I tuned the pwm frequency for the part cooling fans, pid control for the bed is enabled, I'm probably forgetting a couple things.

SebSL83, 2022-09-08:
sounds nice, i stay tuend :)
i dont know your backraound, but maybe you can work togehter with [NARUTOfzr]
or Elegoo to creat great stuff 🥇

Manpoozle, 2022-09-08:
> I'm inclined to think you may have somehow ended up with a corrupted firmware file from here since 1.0.3 has been working for others. I'm glad that fixed it for you :)
> I'm mostly just a hobby coder and have been learning as I go, I helped with the Marlin 2.0.93 configurations for the Neptune 2 series so naturally wanted to port the updated features to the N3 when it came out. NARUTOfzr has been a big help with getting this working so far and hopefully we can continue to make things better. I got another guy that's been trying to get wifi working though there have been some road blocks on that project also but hopefully we can figure something out.

I hear Klipper is pretty nice, but it doesn’t support the touch portion of the touch screen. If someone were to figure that out I think that might be a step up for all of us.

mlee12382, 2022-09-08:
> I hear Klipper is pretty nice, but it doesn’t support the touch portion of the touch screen. If someone were to figure that out I think that might be a step up for all of us.

There's a guy in the Facebook group that is working on configuring Klipper for his N3, he's running it headless (aka no printer screen) and I think he has a pi touch screen instead which is loads better. His current hurdle is getting the strain gauge properly calibrated, the cr6se strain gauge daughter board has a potentiometer on it for adjusting sensitivity but the N3  daughter board unfortunately does not so it's a little trickier to get properly configured it seems, if it were me I'd ditch it in favor of a real probe but he wants to try and figure it out so anyone can do it with the stock hardware.

Manpoozle, 2022-09-08:
I find something elegant about the strain gauge solution, and after years of destabilizing the first generation of hobbyist printers, I now find myself inclined to stay as stock as possible. But kind thanks to those that will walk that path before me. Will a probe set z-height for you? Currently the N3 waits for a feeler gauge and sets that parameter manually for some reason.

I like to load and unload filament from the N3 touchscreen, and bed level. Everything else is remote. I’d be interesting how those two workflows are affected by going headless. I assume the pi screen just shows the pi’s hdmi output? If it was a replacement for the N3’s little touchscreen that might be nice.

mlee12382, 2022-09-08:
> I find something elegant about the strain gauge solution, and after years of destabilizing the first generation of hobbyist printers, I now find myself inclined to stay as stock as possible. But [...]

[image][image][image][image]
 So this is just octoprint with a pi display running octodash, Klipper screen is supposed to be even better  and the fact that you can have the pi / screen mounted next to your printer means you don't really need a screen on the printer itself. 
The strain gauge is a cool concept but imo unless you also have a built in nozzle cleaner and a homing macro that cleans off any residual / leaking filament prior to the final homing and or mesh creation then you're not really getting a consistent measurement by using the nozzle itself as the probe. 
I suspect, though I don't know for sure, that switching to Klipper it would just create the mesh without asking for you to adjust/ set the z offset and that would be handled separately the same way that a bltouch type probe normally works with marlin, though the stock Neptune 2/2S/2D firmware handled it the same way as the 3, updating to marlin 2.0.93 fixed that issue though. 
Here's an interesting thought I had for the N3 with Klipper, add a regular probe and retain the strain gauge also, use the probe for homing and creating the mesh and then after wiping the nozzle on a brush mounted just off the build plate to clean it use the strain gauge to automatically set the z-offset without user interaction similar to the Voron SexBolt upgrade. (Yes that's what they call it and I'm not sure what the back story is, LOL) this would give you the best of both worlds.

Manpoozle, 2022-09-09:
This looks amazing and is incredibly helpful. I really really appreciate it. Do you know why the probe can’t be used to measure the z-offset? And I totally get what you’re saying about the nozzle being a mediocre stand-in. I can imagine a world where the wipe is perfect and with little retraction bursts, the filament is kept at a negative pressure during the whole test. I imagine the cost of an SCT strain gauge is less than the probe, and more manufacturable in less wealthy places. Kinda fits the reprap philosophy. Or I’m just wrong. I’m excited about strain gauges in general for other low-cost high-value consumer machines. The trick would be sending the strain gauge signal back up the stepper-motor wires. That would be the ticket. Perhaps timed between the pulses… that actually might be feasible 🤷‍♂️

mlee12382, 2022-09-09:
> Do you know why the probe can’t be used to measure the z-offset? 

Tbh I'm not 100% sure on the particulars but I think it has something to do with needing multiple reference points to calculate from.

NARUTOfzr, 2022-09-09:
> 

Usually after updating the firmware, the leveling data will be cleared to zero.  If it is not cleared, it may be some unknown bug.  You can select "Restore factory Settings" in the Settings.  At noon, I printed the Gcode file you provided.  Use firmware v1_1.0.3.  Various possibilities have been tried, but the situation you have described has not been reproduced.  But it's good to see that you updated Michael's firmware to address this issue.We will continue to improve firmware bugs.

NARUTOfzr, 2022-09-09:
> > > > Idk if it will fix your problem but try my
> > > 
> > > 
> > > the reflashing of the Screen dosent work. the issues is still the same. i am wondering that the auto level data still be there after reflahing board and screen...
> > > now i will try your 1.0.4 did you change something?
> > 
> > 
> > It's built on the new code from this repo that hadn't been compiled yet, my version does enable some additional features also though.
> 
> so after flashing your version thr printer runs :) thank you could you explain some of your features? at the first view i see nothing new ^^ THX

@SebSL83 Sorry for the late reply!  Usually, after updating the firmware, the leveling data will be cleared to zero.  If it is not cleared, it may be some unknown bug.  You can try selecting "Restore Factory Defaults" in the Settings.  In the afternoon, I printed the Gcode file you provided.  Use firmware V1_1.0.3.  Various possibilities have been tried, but the situation you have described cannot be reproduced.

This reset problem I think is probably some kind of "watchdog" BUG.  Because the BUG is not fixed before the relatively easy to appear this problem, but the firmware V1_1.0.3 was uploaded after the repair, there may be negligence BUG.  But it's good to see that you have updated Michael's firmware to address this issue.

SebSL83, 2022-09-13:
@NARUTOfzr 
thank you generell for replies. i see you finished V1_1.0.4 :)
i will test it soon

SebSL83, 2022-09-15:
Hello again,

today i tested V1_1.0.4 from @NARUTOfzr.
Bedleveling, genrelle settings all works fine.
than i started a new print and get the same issues as i have with V1_1.0.3
In the next step i compiled it my self with VSCODE. After flashing the FW 
i did the same procedere like befor. leveling and so on.
Starting the print and same issues, after homeing, it resets.
Than i flashed the FW from @mlee12382 (V1_1.0.4)
and the Printer runs again without problems.

I dont know what im doing wrong or what problem i have.
the only thing i wondering is that the FW from @NARUTOfzr and the one i compiled have a size from 133kb
the FW from @mlee12382 has 147kb size. imo the size should be near identic or did Michael add mutch more code
as atm is visibile. 

When i can help, with anything, i will do it :)

Thx for reading :)

Manpoozle, 2022-09-15:
Compiling any project from source hasn’t been a great time for me (as a software developer). If you have a normal Neptune 3, I don’t think a debug route that involves compiling new firmware is the best foot forward.

—fly on the wall, Ben

mlee12382, 2022-09-15:
It makes sense that mine is bigger since I have enabled some extra settings, even though they're not visible on the screen yet, I still haven't figured out how to do the screen firmware to make more stuff visible, I have enabled stuff like linear advance and pid for the bed among other things that aren't part of the elegoo firmware.

mlee12382, 2022-09-15:
@SebSL83 my source code is here if you're curious what I've changed:  https://github.com/mlee12382/Neptune_3

SebSL83, 2022-09-15:
> my source code is here if you're curious what I've changed

oh thx, im not realy good at this, but when i have some time i will take a look.
and i can testing compiling your source code and see if it works XD

NARUTOfzr, 2022-09-15:
> Hello again,
>
> today i tested V1_1.0.4 from @NARUTOfzr. Bedleveling, genrelle settings all works fine. than i started a new print and get the same issues as i have with V1_1.0.3 In the next step i [...]

How strange! I really want to figure out why! I will compare Michael's configuration to find the reason.
Can you help with this test? With the firmware I compiled, if there is a restart problem during printing, add the code M413 S0 at the beginning of the gcode file. Try printing again.
Thanks in advance!

mlee12382, 2022-09-15:
> add the code M413 S0 at the beginning of the gcode file. Try printing again.

This makes sense,  I have power loss recovery turned off in my firmware since it has been causing print quality issues for some people. Also @NARUTOfzr I noticed you were working on skew correction but you didn't actually turn it on if you meant to.

SebSL83, 2022-09-16:
> How strange! I really want to figure out why! I will compare Michael's configuration to find the reason. Can you help with this test? With the firmware I compiled, if there is a restart problem [...]

@NARUTOfzr of course, this afternoon i have some time, i will test it and share the results with you :)

SebSL83, 2022-09-16:
@NARUTOfzr 
so i did some tests. When M413 is enable, the problem with the resets at prting start, came every time when starting printing.
Is M413 diabled the print starts normal and everything works fine. I change M413 severall times and the problem reproducible.

NARUTOfzr, 2022-09-17:
> This makes sense, I have power loss recovery turned off in my firmware since it has been causing print quality issues for some people. Also @NARUTOfzr I noticed you were working on skew correction [...]

@mlee12382 It does seem that there are some bugs in the power recovery function.
About the skew correction：Yeah, I didn't turn it on. Because I need to test its practicality, and study its use techniques. Also make sure it doesn't create other bugs.

NARUTOfzr, 2022-09-17:
> @NARUTOfzr so i did some tests. When M413 is enable, the problem with the resets at prting start, came every time when starting printing. Is M413 diabled the print starts normal and everything works fine. I change M413 severall times and the problem reproducible.

@SebSL83 Thank you very much. The results of this experiment confirm some questions for me. I have a general idea of the cause (probably TF card issues) and maybe this can be optimized in firmware, but it will take some time for the software developers.

If you need power recovery function, then you can try to replace the TF card (8G). If there are no more TF cards, I can try compiling new firmware for you to try, if you don't mind (because I'm not sure if it can be solved).

SebSL83, 2022-09-18:
Hi, 

Atm i dont need the recover funktion. So i disabled M413 and all is fine. 
Maybe i switch between your and Michaels Firmware. To Check what Performance better for me 😁 
For future Test i stay tuned here :)