NARUTOfzr/Neptune_3 #4: Is there Linear Advance available

egs2003, 2022-09-14:
Is there Linear Advance available in marlin firmware of neptune 3?

mlee12382, 2022-09-14:
There is on my advanced features build: https://github.com/mlee12382/Elegoo-Neptune-Marlin/tree/bugfix-2.1.x/config/Neptune%203/Advanced%20Features

Manpoozle, 2022-09-14:
Sorry if this is a dumb question, but what’s the difference between the Marlin firmware and the built in Elegoo firmware?

mlee12382, 2022-09-14:
> Sorry if this is a dumb question, but what’s the difference between the Marlin firmware and the built in Elegoo firmware?
> 
In the case of the Neptune 3, nothing except for a few additional features that have been enabled and a couple minor tweaks on settings. For the Neptune 2 /2S/ 2D it is a much newer version of marlin and the interface changes and also the extra features are enabled.

closed 2022-10-03 by egs2003