NARUTOfzr/Neptune_3 #5: Printer resume button doesn't work

Manpoozle, 2022-09-16:
The first time my Neptune 3 failed a print because of a power outage, and the second time was because of a filament jam upstream of the printer. In both cases the printer knew it was in a problem situation and paused with the error screen. The RESUME button on the following screen doesn't seem to do anything. The temperature of the bed and nozzle stays off, no physical parts move, the screen doesn't change, and I can keep pressing the RESUME button over and over with no effect. It's like it isn't wired up to anything. Is this a known issue? 

I put on both the screen firmware and the main board firmware that were available until this most recent main board firmware was released sometime in the last week (haven't tested it).

Thank you, warm regards,
Ben

NARUTOfzr, 2022-09-18:
> The first time my Neptune 3 failed a print because of a power outage, and the second time was because of a filament jam upstream of the printer. In both cases the printer knew it was in a problem [...]

Hi, Ben! Normally, the machine will not be unable to continue printing.
This may be caused by a card reading error. In general, if a card reading error occurs, "ERR" is displayed in the upper right corner of the screen. You can try formatting TF cards and printing small models to test these features. V1_1.0.4 has some optimizations for these features and recommend trying the new firmware.

Manpoozle, 2022-09-18:
What if I’m not printing from the SD card? I’m printing from Octoprint. Is it still able to recover?

Thank you!
Ben

mlee12382, 2022-09-18:
> What if I’m not printing from the SD card? I’m printing from Octoprint. Is it still able to recover? Thank you! Ben

No, Octoprint streams the print gcode one line at a time to the printer, nothing is stored on the printer as far as print instructions go so there is no way for the power loss recovery in the printer firmware to work when it is being controlled by Octoprint or pronterface or direct USB control from any other program/ device. There may be an Octoprint plugin that will function similarly though but I haven't looked so I'm not familiar with them if they exist or how well they actually work.

Manpoozle, 2022-09-18:
Ah, that makes sense. I found a plugin PrintWatch that I’ll try out. It looks like the actual gcode manipulation in order to resume a failed print is pretty straightforward as long as some component is journaling what’s going out. I could probably write it.

Alternatively, I’ve seen some funny looking boards that hang out of the SD card slot and allow for high throughput data transfer, which can open up that as still being remote connection. Not sure if you’ve tried one of those?

Regards,
Ben

mlee12382, 2022-09-18:
> Alternatively, I’ve seen some funny looking boards that hang out of the SD card slot and allow for high throughput data transfer, which can open up that as still being remote connection. Not sure if you’ve tried one of those?

I have not seen or heard of those. The couple of times I've tried using power loss recovery the resumed portion of the print failed in one way or another so I don't even bother with it anymore. Unless you're printing expensive filaments or multiple day prints that you don't have time to start over or print the rest separately and glue them I don't really see the benefit personally.

Manpoozle, 2022-09-18:
That makes sense. I can’t speak to the usual success rate, but with these two situations, I would have given it a shot.