NARUTOfzr/Neptune_3 #7: Z offset

RickzFPV, 2022-09-19:
I am having a big issue with my Z offset changing constantly, printer will print 1 good print then I do wat the instructions for the firmware says, press confirm then go into temperature settings, then level settings, then save.  Done this many times just trying to get the z offset to be the same for 2 consecutive prints with Zero luck.  I am running the 1.0.4 firmware.  Any help with this would be very much appreciated, I really do like and want to use the printer, when it is printing it prints great.

Thank You
Rick

Manpoozle, 2022-09-19:
> When you are bed leveling (tramming), do you have filament in the nozzle? I extrude, and clean the nozzle with a silicon spatula, the retract a bunch (like 100mm), then bed level with a piece of paper (should feel some resistance from the nozzle on the paper, usually a couple steps down from where you first feel the paper, but not far). Then I extrude again until the filament is coming out once again. Clean it off and begin printing.
>
> Also I clean the bed with iso between every print as the last step after pulling up all the plastic, and it makes a world of difference. They have these little bottles that pump a little iso up to the top when you press on them, reduces the time to clean it down to like 10 seconds, so doing it every time matters not.
>
> Hope this helps,
> Ben

On this note, does anyone know how to disable the extra z-offset check the printer does in the center of the build plate at the beginning of the print? It leaves a little bit of goo that ruins the bottom of the print surface when printing in the center of the build plate.

Oh and Rickz, I’ve heard that manually leveling (again, tramming) the bed with the physical hardware before having the printer find the offset grid helps a lot. And if you need more help these folks will probably need more detail as to how your print is failing…

Regards,
Ben

RickzFPV, 2022-09-19:
I am still having the same problem.  I can set zoffset and get 1 good print, then when I try to run a 2nd print the nozzle is too high.  that causes print to fail due to adhesion.  I know my gantry is dead on level and even with the bed surface that is always one of the first things I check.  I will post a picture of the screen this time around when I am done again, because there are some variances in the bed that don't seem right to me.  It printed great till I updated the firmware then this problem began

Manpoozle, 2022-09-19:
> If you're using cura replace your start gcode with the following and the hotend won't heat up enough for the filament to melt until it's at the front corner for the prime line, if you use the newest prusaslicer or superslicer the included N3 profiles already act this way.
> G90 ; use absolute coordinates
> M83 ; extruder relative mode
> M104 S120 ; set temporary nozzle temp to prevent oozing during homing and auto bed leveling
> M140 S{material_bed_temperature_layer_0} ; set final bed temp
> G4 S10 ; allow partial nozzle warmup
> G28 ; home all axis
> ;G29 ; run abl mesh (add-remove ; at the beginning of line to disable-enable)
> M420 S1 Z5.0 ; load mesh
> G1 Z20 F240 ; raise nozzle
> G1 X2 Y10 F3000 ; move nozzle into position for priming
> M104 S{material_print_temperature} ; set final nozzle temp
> M190 S{material_bed_temperature_layer_0} ; wait for bed temp to stabilize
> M109 S{material_print_temperature} ; wait for nozzle temp to stabilize
> G1 Z0.28 F240 ; move nozzle down to prime position
> G92 E0 ; reset extruder
> G1 Y140 E10 F1500 ; prime the nozzle
> G1 X2.3 F5000 ; move nozzle over for 2nd prime line
> G1 Y10 E10 F1200 ; prime the nozzle

Michael is that replacement startup gcode?

mlee12382, 2022-09-19:
> Michael is that replacement startup gcode?

Yes if you are using cura

mlee12382, 2022-09-20:
I'm not sure what the issue is with the z offset changing, have you tried my advanced features firmware to see if it makes a difference? It may not since it's built on the same source code at the moment with other features enabled, but there's a slight chance it might improve your experience. 

As for the Neptune X I have attempted to get Marlin built for it but it seems to get stuck on 100% during the flashing process, you can try it if you want though and if it doesn't work just flash back to stock firmware from the elegoo website. 

https://github.com/mlee12382/Elegoo-Neptune-Marlin/tree/bugfix-2.1.x/config/

RickzFPV, 2022-09-20:
@mlee12382 thank you I will try and flash the advanced features firmware.  Do I need to flash a new Dwin also or just the machine firmware

mlee12382, 2022-09-20:
> Do I need to flash a new Dwin also or just the machine firmware

If you are running the current dwin_set then it is the same and you should be fine with just flashing the board.

RickzFPV, 2022-09-20:
@mlee12382 how do I download your advanced features firmware, I seem to have the hardest time with github if it doesn't have a zip file to download

mlee12382, 2022-09-20:
> @mlee12382 how do I download your advanced features firmware, I seem to have the hardest time with github if it doesn't have a zip file to download

If you click on the .bin it will give you an option to download it.

mlee12382, 2022-09-20:
> @mlee12382 how do I download your advanced features firmware, I seem to have the hardest time with github if it doesn't have a zip file to download

[image] this is from my phone but should be similar on a computer.

RickzFPV, 2022-09-20:
@mlee12382 Mike your firmware didn't work on my printer, after flashing it just stays on the elegoo screen.  I went back to the 1.0.4 and everything is fine.  got a question I noticed on mine that some of the screws were a little loose on the bed so I tightened everything down, curious though if there is a way to tram or level the bed manually to get it as close as I can to perfectly level.

RickzFPV, 2022-09-20:
Yeah Michael everything is squared on the frame triple checked lol.  But my wire harness is zip tied to the bottom Eden tube it came that way I will take them off and see if it changes anything

mlee12382, 2022-09-21:
Interesting, I wonder what changed that was causing the offset issue.. 🤔  and I'm not 100% if that will work or not, I think I heard there's a way to turn it off on the touch screen? Also try unplugging it and see what happens.

RickzFPV, 2022-09-21:
@mlee12382 ok I will give it a try by unplugging it and I will look around and see if there is a way yo turn it off.  As far as what changed in the firmware I have no Idea I am still very happy with the printer and I know it is just gonna take a little time and then there will be a full blown firmware for it, it is just so new it will take a little time

SebSL83, 2022-09-23:
Hello again, i think the Z offset procedure has an issues. I have test with both V1.0.4 Firmware, narutos and mlees, i start
the auto leveling. when its done i adjust the Z offset with an paper, so i feel a little resistance. Than i start a print and at the right side of the bed the nozzle scrached the bed. i adjust the offset while printig, and than on the left side, its to high.
for my opinion, that is wrong. when auto leveling is done and the zoffset is adjust, the FW should know the grid and do
mathematics to calculate the perfect height on any point on the bed. (i dont know the detail algorithm)
all my screws are lockt, the pie plate is flat. i see in terminal that the leveling is done and saved.
at the picture you see what i mean, i hope ^^
[image]
[image]
[image]

NARUTOfzr, 2022-09-24:
> Hello again, i think the Z offset procedure has an issues. I have test with both V1.0.4 Firmware, narutos and mlees, i start the auto leveling. when its done i adjust the Z offset with an paper, so [...]

Hi!You can try debugging the machine according to this document.
[Neptune3-Leveling Issue Solutions.docx](https://github.com/NARUTOfzr/Neptune_3/files/9637658/Neptune3-Leveling.Issue.Solutions.docx)

RickzFPV, 2022-09-24:
@NARUTOfzr so I did what was recommended I know now my eccentric nuts were way to tight just set everything back to factory settings, and gonna try it and see how it does.  My question though now is why is the right side of the bed so far off from the left side here is a picture is there a way to remedy this hardware wise vs software.  
[image]

SebSL83, 2022-09-26:
Hi! You can try to debug the machine according to this document.
[Neptune3-Leveling Issue Solutions.docx](https://github.com/NARUTOfzr/Neptune_3/files/9637952/Neptune3-Leveling.Issue.Solutions.docx)

Such a little feedback, I readjusted the V-Wheels as per the Neptune3-Leveling Issue Solutions.docx
it looks like my v-wheels are too tight as well, now my bed grid is pretty good. and after 4 prints in a row it looks very good with the Z offset.

RichBr1, 2022-09-27:
Found this document above a couple of days ago and loosening the rollers so they just bind cured the problem. Messed around for a long time with firmware before with no luck.  When you update the firmware the version on the screen does not change from 1.0.2 so another bug. I think that 1.0.4 is still beta looking at the comments in the download. So glad to have found this site - it all makes sense now.

mlee12382, 2022-09-27:
> When you update the firmware the version on the screen does not change from 1.0.2 so another bug.

Make sure your sd card is formatted fat32 with 4096(may not be the default size) I've heard this may solve the issue. If not there may indeed be a bug but try that.

RichBr1, 2022-09-27:
Thanks, tried this on 2 different cards but sadly it appears to be a bug.

RickzFPV, 2022-09-29:
@mlee12382 @NARUTOfzr Just wanted to let you guys know I used that leveling issues Dox file and all my issues with the Zoffset went away.  Elegoo is shipping the printers with the wheels tight on purpose so nothing shifts around during shipping.  I wish they would put a supplement sheet or something in the instructions about it.  I am back on 1.0.4 firmware and haven't had to change anything, been going for about 3 days now with no issues.  I don't think that the version display is a bug, it is either the sd card or not initializing the eeprom after firmware update.

RichBr1, 2022-09-29:
Just to say I followed the adjustments to the wheels as suggested. While this worked well for a while I had an issue with the extruder wheels loosening and also some of the other wheels not keeping their position and becoming loose or tight over time. Not sure the nuts are eccentric like other printers. It seems like the locknuts or adjustment nut are not holding but will investigate further. I have tried multiple updates to board firmware on multiple SD cards but version sticks stubbornly to 1.0.2 in screen, suspect this is a eeprom initialisation bug after update.

RichBr1, 2022-10-01:
Finally sorted, had to adjust a couple of locknuts as well to get this properly setup - adjusted eccentric nuts to slackest posution with wheel loose and then adjusted locknut gap so that it is just tight enough to not impact wheel rotation. Adjusting eccentric nut will then tighten correctly. Also foung undoing the 2 rear hex screws as in the doc means you have to make sure that the horizontal bar is level with the top frame when tightening. Now have consistent bed levelling and z offset,  Firmware version on scrren still a bug/issue.

RickzFPV, 2022-10-01:
Yeah since I did that de bugging supplement mine has been printing great.

NARUTOfzr, 2022-10-02:
> Finally sorted, had to adjust a couple of locknuts as well to get this properly setup - adjusted eccentric nuts to slackest posution with wheel loose and then adjusted locknut gap so that it is just [...]

Can you film the process of upgrading the screen firmware?

NARUTOfzr, 2022-10-02:
> @mlee12382 @NARUTOfzr Just wanted to let you guys know I used that leveling issues Dox file and all my issues with the Zoffset went away. Elegoo is shipping the printers with the wheels tight on [...]

@RickzFPV 
Thank you for your positive response and feedback.
[image]

According to the previous picture, there is a big difference in the leveling value. You can try to adjust the fixing screws of the hot bed platform (15 screws). In fact, I think the hot bed platform only needs 4 support columns, because for a 235*235 hot bed platform, too many support columns (15 screws) will only reduce the accuracy of the hot bed platform or increase the difficulty of debugging.

RichBr1, 2022-10-02:
Upgrade process
Download main.zip from site and extract screen and mainboard firmware onto sd card
Insert sd in screen (formatted fat32 4096)
[image]
[image]
Remove sd and reboot
put in sd card and reboot
Mainboard firmware updates
[image]
Turn off and on
Check firmware on screen settings
[image]
Firmware used for update 1.0.3
Update file says ZNP_ROBIN_NANO.bin
This is updated to ZNP_ROBIN_NANO.cur after the upgrade
Wonder if the screen firmware files are the latest in main.zip? 
According to version V1 the settings / about should be:
In 【Settings】→【About】, you can view the screen firmware version and the motherboard firmware version?

NARUTOfzr, 2022-10-02:
@RichBr1
Currently, only V1 screen firmware is available.
In your case, the screen firmware did not load properly and needs to be reloaded (normal screen firmware loading time is about 90 seconds).

From the picture you gave us feedback, there are two possible reasons:
1. The memory card is not formatted properly (or there may be a problem with the TF card, which does not support formatting to the required format. You can try to replace another 8G-TF card)
2. There are no files in the [DWIN_SET] folder under the root directory of TF Card or the files are garbled when decompressed. The normal file is as follows:
[image]

RichBr1, 2022-10-02:
Thaanks, wiped and reformatted card 3 times, copied files across and it finally worked and is showing 1.0.3. Have not needed to level again so I think it is now all setup OK.
Is 1.0.4 safe to use yet or isit still beta?
Thanks again

NARUTOfzr, 2022-10-03:
> Thaanks, wiped and reformatted card 3 times, copied files across and it finally worked and is showing 1.0.3. Have not needed to level again so I think it is now all setup OK. Is 1.0.4 safe to use yet or isit still beta? Thanks again

Details of bug fixes and firmware updates are updated to the update log.
The new version of the firmware will fix some of the issues found in the previous version of the firmware, so the new version will be more stable.

RickzFPV, 2022-11-08:
@mlee12382 My Neptune 3 has been printing great on the 1.04 firmware with pretty much Zero issues.  I have a couple questions though.  Is there a firmware out there yet that has an advanced configurations menu for pid tuning and setting esteps on the printer, or if you can let me know what the Baud rate is for it I can go into it through pronterface and do it that way.  I am a little afraid to do too much with it because it is printing so well right now, but would like to do a little tuning on it.  Also Mike down the road if I want to could the board be swapped out for lets say a BTT e3 v3, just not sure if it would mess up all the auto leveling and stuff on it.  Thank you Mike for all your help

mlee12382, 2022-11-08:
> @mlee12382 My Neptune 3 has been printing great on the 1.04 firmware with pretty much Zero issues.  I have a couple questions though.  Is there a firmware out there yet that has an advanced [...]

The screen firmware mods are still in progress, I have a couple of friends who have been working on it off and on so hopefully at some point we will be able to have some of those options on the screen.
The baudrate is 115200.
As for the SKR Mini it will work with the exception that it will also need a different screen firmware or you would need to replace the screen, you can get either an ender 3 style screen which would probably be less headache or you can get the commercial mks h43 which is mostly the same as the stock N3 screen. As far as I know as long as the configuration is set right you shouldn't have any issues with the rest of the hardware. Might be better off switching to a regular probe though just for easy of use.

RickzFPV, 2023-05-04:
Just curious if there will be any more updates for the Neptune 3.  Would really like to see an update to the screen firmware, to be able to have advanced features so it could run the New 2.1.2 firmware.  I am still at times having issues with the Zoffset moving.  I am not sure if bed is warped, because I don’t use it a ton, or if the bed needs to be shimmed to try and make it more accurate.  I guess I need to set it up with octoprint and run the bed visualizer and see what it looks like.  Just wondering if it is kind of dead for this printer because of the new pro printers

geoffreygao1, 2023-08-06:
I believe I am facing the same issue with a changing Z offset. From day to day (without touching the system), the Z offset varies as much as 0.3mm. I am running pre-heat on the system to account for thermal expansion and the nozzle is clear of any extruded material prior to auto bed level. I am not sure if this is a mechanical issue (lost steps in Z?) or firmware

RickzFPV, 2023-08-06:
Mine doesnt do it anymore since i re shimmed the bed now that it is perfectly level it has been printing great i had 1 corner that was way lower than the rest it came that way from the factory it took some time to shim it but it was worth the time

geoffreygao1, 2023-08-06:
Interesting, I will have to try that out. Still don't understand why a corner would cause the center Z-offset value to change every time?

RickzFPV, 2023-08-06:
If the bed is off even on 1 corner it will also push the center out