NARUTOfzr/Neptune_3 #11: WIFI Module Support

EastsideCustoms93, 2022-09-26:
I am using the Neptune 3 for about 5 weeks now. It works so far, but for more convenience I would like to have the WIFI Module working as it seems to do on the Neptune 2 etc. . 
I already installed the hardware and the small LED flashes at boot shortly. So theoretically it should work with the appropriate software.

Can this be added, please? 

Thanks  for your work!

RichBr1, 2022-09-27:
Try Octoprint - works great on a pi plugged into the printer and has video etc. You can wifi or ethernet to the machine.

EastsideCustoms93, 2022-09-27:
@RichBr1 I use Octoprint on my other printers. But one: nowadays a Pi is quite expensive and two: the mainboard has the option, so why not using it!!!

Nevertheless,
 thank you for your help

danielk117, 2023-01-17:
Wifi support would be great. Why using external hardware, if all the stuff could be done on the printer itself. 

I installed the MKS Wifi board too... Hoping that support will come soon 😉

mlee12382, 2023-01-17:
> Why using external hardware, if all the stuff could be done on the printer itself. 
> 
The mks wifi module even if it worked with the N3 only gives VERY basic functionality. You can transfer files and move the print head and start / pause/ cancel prints. You might also be able to do some terminal stuff. It does not connect outside of your home network, there's no way to add cameras or do any of the more advanced features / plugins. Adding external hardware like a Beagle or Octoprint or Klipper is FAR more advanced and feature inclusive than you can ever get by just adding a wifi chip to the printer. The only real benefit of the chip is that you don't have to move the sd card back and forth and that's about it.

danielk117, 2023-01-17:
> The mks wifi module even if it worked with the N3 only gives VERY basic functionality. You can transfer files and move the print head and start / pause/ cancel prints.

Thats exeactly what I want. I basically doesn't do more with octoprint. (I already have octoprint running, but I want to place my printer to another room and the usb connection is a problem.)

> It does not connect outside of your home network

Great. I hate devices which connects to someones cloud service 😉 

> there's no way to add cameras or do any of the more advanced features

My plan is to use [this mod](https://github.com/xreef/MKS_WIFI_upgrade_with_BeePrint_web_interface) or [the cura plugin](https://github.com/Jeredian/mks-wifi-plugin).

@mlee12382 
I see you have an customized fork of this repo. So adding...

```
  #define SERIAL_PORT_2 2
  #define BAUDRATE_2 115200 
  #define MKS_WIFI_MODULE
  #define COMMUNICATION_PROTOCOL_MKS_SERIAL
```

...to `Configuration.h` and building it, could do it?

mlee12382, 2023-01-17:
@danielk117 the limited testing I've done with the help of a friend who's more knowledgeable than I am hasn't produced positive results yet. The module will power on and you can access it's web server in AP mode but it doesn't actually communicate with the printer or anything. It's basically just working as if it wasn't even connected to anything but power and ground.

mlee12382, 2023-01-17:
@danielk117 https://youtu.be/nHNZPRl8gzA this option might interest you.

danielk117, 2023-01-17:
> The module will power on and you can access it's web server in AP mode but it doesn't actually communicate with the printer or anything. It's basically just working as if it wasn't even connected to anything but power and ground.

Exactly the same behavior as the stock firmware.

> https://youtu.be/nHNZPRl8gzA this option might interest you.

A bit too limited, but nice idea. 😉

danielk117, 2023-06-28:
I can report some success. I bought the MKS WIFI module, which is a simple ESP chip and fits directly on the MKS board. I flashed it with [ESP3D v3](https://github.com/luc-github/ESP3D/tree/3.0), which provides a simple web interface to monitor and control your printer using gcode commands.

ESP3D v3 supports the mks serial protocol, but I wasn't able to active it in the marlin firmware (set `#define SERIAL_PORT_2` and/or `#define MKS_WIFI_MODULE` is not enough). So I let ESP3D using its standard `RAW_SERIAL`.

In my marlin `configuration.h` I set the following options, to enable the serial ports for the wifi module:

```
#define SERIAL_PORT 2
#define SERIAL_PORT_2 1
```

Thats it. ESP3D provides an AP and you can configure the access to your local network. Than you can easiely access the web interface.

If I'm correct, ESP3D also provides a serial2net solution, so you can run Octoprint and the [OctoPrint-Network-Printing-Plugin](https://github.com/hellerbarde/OctoPrint-Network-Printing) (see [esp-01-octoprint](https://github.com/R0peE/esp-01-octoprint)). But I havn't tried it yet.