NARUTOfzr/Neptune_3 #8: Falsely reported thermal runaway message on lcd screen

SpeCter, 2022-09-26:
Right now it seems like there might be a small mistake in the firmware.

In modules/temperature.cpp there are two conditional checks which would trigger a thermal runaway shutdown while doing PID tuning.
This works in the case of an actual thermal runaway but still might trigger the on screen message on the lcd screen even when there is no thermal runaway due to missing braces.

The conditionals which would need proper bracing would be in line 742 and line 746 otherwise ScreenHandler.GotoScreen(MKSLCD_Screen_ERR6_POPUP);
and
ScreenHandler.GotoScreen(MKSLCD_Screen_ERR7_POPUP);
might be called even when the condition isn't met for thermal runaway under certain conditions.

This leads to the screen falsely reporting thermal runaway while the printer is still functioning normaly(can still be controlled by pronterface/octoprint or similar terminals and will finish the current pid tuning without problems) except for the screen which can't be recovered until you restart.

NARUTOfzr, 2022-09-26:
Yes, this definition is not very reasonable. It's easy to misunderstand.
Because it is not recommended to do other operations when PID autotune. I want to define it as: GotoScreen(MKSLCD_SCREEN_HOME);

skwp, 2022-10-03:
Hey @SpeCter sorry for reaching out here, but I cannot find a way to DM you. Can you please contact me on twitter at @skwp or yan at swan dot com - have something to discuss. Thanks.