NARUTOfzr/Neptune_3 #9: M420 S1 neccessary

egs2003, 2022-09-26:
Is the command M420 S1 neccessary to loud bed leveling mesh in v1.0.4?

NARUTOfzr, 2022-09-26:
Unnecessary. Leveling is enabled after G28 by default. Default fading height is 50mm.

NARUTOfzr, 2022-09-26:
#10 If there is a leveling problem, please refer to the scheme for debugging.

egs2003, 2022-09-27:
The hardware is completely set. The leveling also brings reproducible results. However, when printing over a large area, I always have strong layer fluctuations in the same place.

[image]
[image]

NARUTOfzr, 2022-10-02:
@egs2003 Make sure the training wheels turn easily? Try releasing the training wheels and leveling them again.

egs2003, 2022-10-02:
what exactly is meant by "turn easily". I have set them so that they have secure contact with the profile, but can be rotated by hand when the axis is fixed.