NARUTOfzr/Neptune_3 #12: pause/resume not working (printing from SD card)

bsas, 2022-09-29:
My Neptune 3 has a very weird behavior (bug): when I pause the print and resume, looks like the extruder gets crazy and stops extruding (looks like it actually starts to move the step motor backwards). Is it just my unit? I tried to update the firmware to the 1.0.4 and still the same issue.

bsas, 2022-09-30:
Here is an example of the issue: the print starts perfectly and the extruder is working as intended (video: [https://youtube.com/shorts/zil7fMgexjc](https://youtube.com/shorts/zil7fMgexjc?fbclid=IwAR0wPsPtmh6xZwhslNOBNwCkxDXdZlCsewYRGoXbVcCZKp7r5zVwHwzUw3c)) but then, as soon as I just pause and resume, the print fails and the extruders starts to glitch out (video: [https://youtu.be/WFoejn8GAos](https://youtu.be/WFoejn8GAos?fbclid=IwAR0Sg00bL6x3wXxQO7GG8HzUtc_B3JzYXeAIRMHNv1S9ikwjEot0skhEWo0)).

NARUTOfzr, 2022-10-05:
@bsas 
Hey, sorry I'm late. From the video, the extruder appears to be working properly. The operation continues to print and the machine action is normal.

I would like to ask whether the whole model can be printed normally without suspending the printing. What is the quality of the printed model?

I think the reason for the extrusion failure after the pause: it may be the clogged nozzle. You can try increasing the print temperature/changing the nozzle/changing the filament. Also, please check whether the cooling fan of the print head is normal (as shown in the picture)?
[image]

bsas, 2022-10-05:
Hello,

No, I am pretty sure the extruder is not working after resume.

The head goes back and moves like it is printing and the heat of the nozzle just melts the top layer, then, since there is no extrusion happening it starts to print in the air after a Z change. If I leave it printing it is for sure extruding nothing. It keeps moving in the air like it is printing, but, no plastic comes out.

If you observe my videos carefully, in the beginning of the print (while it is working fine) the extruder gear is moving back and forward (as expected). But, after the resume, it starts to "click" and it just moves backwards. If I leave it eventually it spits the filament out because it is just moving backwards and clicking, so, it is slowly removing the filament from the tube.

I am pretty sure that the issue is NOT on the main board firmware and it is not a clog because if I print via USB on something like Octoprint it pauses and resumes perfectly. It is some bug in the "screen" code.

bsas, 2022-10-08:
So... still not working. Any updates on that? I am pretty sure the issue is on the screen FW, not the board FW.

NARUTOfzr, 2022-10-09:
@bsas 
I'm sorry! I don't have much time to deal with problems here on weekdays. I will feed the problem back to software engineering and see if they have any solutions.
I have some ideas about the problem, but I'm not sure. I will compile a firmware for you to try tomorrow morning.

NARUTOfzr, 2022-10-09:
Please try updating this firmware：
[Neptune3_test.zip](https://github.com/NARUTOfzr/Neptune_3/files/9742055/Neptune3_test.zip)

bsas, 2022-10-10:
YES! That totally solved the problem!

One side effect tough is that now when you press "PAUSE" looks like the printer flushes it's buffer. It prints some more lines before it actually pauses... Don't remember if that was the previous behavior tough :(

mlee12382, 2022-10-10:
@NARUTOfzr what did you change to get the resume to work?

NARUTOfzr, 2022-10-12:
> YES! That totally solved the problem!
> 
> One side effect tough is that now when you press "PAUSE" looks like the printer flushes it's buffer. It prints some more lines before it actually pauses... Don't remember if that was the previous behavior tough :(

The printer will run all the lines in the buffer before executing the pause action. I think there should be a better way to pause, but it might be complicated, and the marlin community firmware seems to have a similar problem.

NARUTOfzr, 2022-10-12:
[image]
@mlee12382 I annotated the gcode that was retracted after the pause and the gcode that was continued. Possibly because the retraction distance after pause is too long (10mm), some filaments may expand and get stuck in the gap of the print head.
There may also be some unknown bug, but I haven't encountered this problem in many trials.

parkerbol, 2022-11-27:
Hi I have the issue that after pause for filament change for example, the printer resumes 1 ou 2 layers too high or badly at left or right. I am in 1.0.5beta

Signum, 2022-11-27:
Yes, same here. After a pause the print is ruined.

Herbert-Testmann, 2023-02-21:
my neptune3 is on 1.0.5 Beta and I have the same problem after Pause.
The filament goes backward no forward.
If the bug is fixed in the "test FW" from Okt/9/2022 - when will we get it in a new Beta?