NARUTOfzr/Neptune_3 #13: Full access on octopi for Neptune 3

ruuckythe1st, 2022-10-02:
I would like to have the full control and feature for the Neptune 3 on the octopi. The infobox from octopi say: 

"Your printer's firmware supports host action commands, but they are disabled. The firmware will not inform OctoPrint about job pause & cancellations, filament runouts, and similar firmware-side events If you want this functionality, you need to
change your firmware's configuration."

My question is: 
- How i can activate the host action commands for the Neptune 3?

- What do you have to do for it?

Moody66, 2022-10-02:
https://www.youtube.com/watch?v=YDvFN51siMU

ruuckythe1st, 2022-10-02:
I can't build the Marlin ZNP_F401_Marlin-V_1.0.4 from NARUTOfzr in Marlin Firmware Auto Build
[image]

I have no idea How i can result this problem...

mlee12382, 2022-10-02:
You need the full set of files from this repo, other things have been added and modified beyond the standard configuration_adv.h and configuration.h files and it won't work without them. You also should use the platformio build tool and not auto build marlin. You can get pre-compiled firmware with host action commands and other things enabled here: https://github.com/mlee12382/Elegoo-Neptune-Marlin/tree/bugfix-2.1.x/config/Neptune%203/Advanced%20Features  no nozzle preheat seems to work best for people.

ruuckythe1st, 2022-10-02:
Thanks for that helpful information, but it looks like it is not official. How stable is that version from [mlee12382](https://github.com/mlee12382/Elegoo-Neptune-Marlin/commits?author=mlee12382)?

Maybe **NARUTOfrz** can tell when an official stable release is coming?

Thanks for helps!

mlee12382, 2022-10-02:
It is as stable as the official firmware as it's built on the official firmware, basically the same thing you are trying to do yourself. I've enabled Host Action Commands, Linear Advance, PID for the bed, turned off the broken power loss recovery, set the fade height at a much more reasonable 5mm instead of 50mm, and several other minor improvements that I can't remember off the top of my head. Any bugs that may still be present in the official firmware are likely still present, some people have reported better luck with my firmware than stock. My fork of this repo has the config files, I uploaded the compiled firmware to my main Neptune Marlin repo though so it's all together in one place.

ruuckythe1st, 2022-10-02:
ok, thanks! I'll try it and will let you know if it works.

mlee12382, 2022-10-02:
One thing to note is that the filament run-out sensor is handled by the screen firmware and not by the main board firmware so it may or may not report status to octoprint, I haven't had anyone tell me for sure one way or the other.

ruuckythe1st, 2022-10-02:
Ah I understand :)
This is because of the two firmware. One for the mainboard and the other for the touchscreen. The touchscreen does not pass the information on to the mainboard.

ruuckythe1st, 2022-10-04:
i must just install from you the No Nozzle Preheat -> ZNP_ROBIN_NANO.bin
is this right?

ruuckythe1st, 2022-10-04:
But you have just activate the host action commands in this ZNP_ROBIN_NANO.bin?

mlee12382, 2022-10-04:
> But you have just activate the host action commands in this ZNP_ROBIN_NANO.bin?

Host Action Commands, Linear Advance, turned off the buggy Power Loss Recovery, Fade Height is set to 5 instead of 50, enabled M600 though it seems to have trouble with the resume after filament change which I suspect is an incompatibility with the screen firmware,   I properly defined the drivers, and some other minor tweaks I can't remember off hand.

closed 2022-10-04 by ruuckythe1st