NARUTOfzr/Neptune_3 #24: Temperature with leveling

egs2003, 2022-10-06:
@mlee12382

The presets for the temperature are overwritten during leveling. For example, the preset temperature is 45°C. It heats up to this at startup, but then changes to 65°C.

        There is on my advanced features build: https://github.com/mlee12382/Elegoo-Neptune-Marlin/tree/bugfix-2.1.x/config/Neptune%203/Advanced%20Features

_Originally posted by @mlee12382 in https://github.com/NARUTOfzr/Neptune_3/issues/4#issuecomment-1247040144_

egs2003, 2022-10-06:
I print at 40°C bed temperature. Therefore, I level at slightly above the target temperature.

mlee12382, 2022-10-06:
Ah ok, you're not having issues with prints sticking with the bed that low?

egs2003, 2022-10-06:
Yes, I already had. I use the Fillament NX2 from extrudr. The bed adhesion has caused big problems. For me, the solution was to start printing with 40°C bed temperature and 230°C nozzle. After the first layer I increase to 50°C and lower the Nozzle temperature to 200°C. This rarely causes any problems.

mlee12382, 2022-10-07:
@egs2003 https://github.com/mlee12382/Neptune_3 Here you go, I updated the preheat settings so it should only rely on the ABL preheat settings in the menu on the screen, it also has the updates from the newest beta 1.0.5 firmware, if you have any issues let me know :) There is a blue and green theme available, not 100% sure the blue will actually work, green is what has been available up until now but I finally got the blue to compile.

egs2003, 2022-10-07:
What does blue and green theme mean. You have only provided the ZNP_ROBIN_NANO.bin. Themes are screen firmwares I thought?
The leveling temperature is now taken over correctly.

egs2003, 2022-10-07:
The fans are not spinning anymore!

mlee12382, 2022-10-07:
I believe that the screen firmware in theory has the capability for both colors and the board firmware can tell it which to use, I have not confirmed this though. 

As for the fans and the pause that is annoying, possible it's a glitch is 1.0.5, I will update 1.0.4 this evening and upload it for you since that didn't work :( reflash from my other link to go back so it's at least useable.

egs2003, 2022-10-07:
I flashed the green theme. But the screen was still blue.

mlee12382, 2022-10-07:
> I flashed the green theme. But the screen was still blue. 

Ahh ok, that's a bummer but good to know, there must be something in the screen firmware that controls it or has it disabled then. 
Which fan or fans aren't working? Just the hotend? Or the part cooling fans?

egs2003, 2022-10-07:
Definitely the hot end cooler. I'm not so sure about the part coolers at the moment. I have already flashed back

mlee12382, 2022-10-07:
> Definitely the hot end cooler. I'm not so sure about the part coolers at the moment. I have already flashed back

Alright I will work on it after I get home from work.

mlee12382, 2022-10-07:
@egs2003 🤦 I set up the hotend fan by calling FAN1_PIN which is normal but for some reason the FAN1_PIN definition in the pins file was commented out so that's why it wasn't working, it's corrected now and I'm recompiling and will upload shortly.  Lol at least it was a simple fix.

mlee12382, 2022-10-08:
@egs2003 https://github.com/mlee12382/Neptune_3/tree/main/Advanced%20Firmware here you go, try this out and see if it fixes your issues. Let me know how it works.

egs2003, 2022-10-08:
I I won't be able to test it for another week. Thanks for the quick implementation

egs2003, 2022-10-18:
It's working

closed 2022-10-18 by egs2003