NARUTOfzr/Neptune_3 #30: Resume print paused using M0 for filament change

niklash-dev, 2022-10-16:
It seems like it is currently not supported to resume a paused print for filament change using Gcode (M0). Would be great if this support could be added,

I guess multiple people are searching for this and there is currently no reliable solution:

[https://www.reddit.com/r/ElegooNeptune3/comments/xerp71/changing_filament_mid_print/](https://www.reddit.com/r/ElegooNeptune3/comments/xerp71/changing_filament_mid_print/)

[https://www.reddit.com/r/ElegooNeptune3/comments/y2ejix/pause_at_layer_for_filament_chage/](https://www.reddit.com/r/ElegooNeptune3/comments/y2ejix/pause_at_layer_for_filament_chage/)

NARUTOfzr, 2022-11-23:
@niklash-dev M0 is supported by V1_1.0.5_Beta firmware.

niklash-dev, 2023-08-18:
@NARUTOfzr Thank you, I will close this issue now.

closed 2023-08-18 by niklash-dev