NARUTOfzr/Neptune_3 #37: Bed tilt correction (experimental) : Application & instructions

NARUTOfzr, 2022-11-18:
Tutorial in progress...

abge571, 2022-12-06:
Please, I have a problem with the tilt of my bed and I have no idea how to fix it.

NARUTOfzr, 2022-12-06:
Can you take pictures of the leveling value?

abge571, 2022-12-06:
> Can you take pictures of the leveling value?
> […](#)

Yes, here you have them, I add my printer it worked fine until 1 week ago, I've been with it for 2 months and I don't understand the problem, I tried updating the firmware and adjusting the eccentrics but it didn't work, it keeps printing badly on the same side

[image]

NARUTOfzr, 2022-12-06:
如果您正在使用固件版本V1_1.0.5-Beta。你可以尝试在起始代码中加入"M853 L-0.2 M-0.2 "

abge571, 2022-12-06:
> 

Yes, I use that version, but how to add the code?

abge571, 2022-12-06:
> 如果您正在使用固件版本V1_1.0.5-Beta。你可以尝试在起始代码中加入"M853 L-0.2 M-0.2 " 解压这个文件。打印“EN3_grid test3.gcode”我已经在这个文件中加入这段代码。 [EN3_grid test3.zip](https://github.com/NARUTOfzr/Neptune_3/files/10161770/EN3_grid.test3.zip) [image]
> 
> 在打印中观察1234的位置：如果23位置仍然被压得很紧，则尝试继续减小“L&M数值”。您可以将图片发给我看看。 [image]

OK I will try it

abge571, 2022-12-06:
> 如果您正在使用固件版本V1_1.0.5-Beta。你可以尝试在起始代码中加入"M853 L-0.2 M-0.2 " 解压这个文件。打印“EN3_grid test3.gcode”我已经在这个文件中加入这段代码。 [EN3_grid test3.zip](https://github.com/NARUTOfzr/Neptune_3/files/10161770/EN3_grid.test3.zip) [image]
> 
> 在打印中观察1234的位置：如果23位置仍然被压得很紧，则尝试继续减小“L&M数值”。您可以将图片发给我看看。 [image]

Now the part on the left prints low and the part on the right prints well, should I lower the -0.2 of L? If so, what number?

abge571, 2022-12-06:
> @abge571 尝试L-0.3 M-0.3
> 
> 抱歉，搞混了。尝试增加L&M，L-0.15 M-0.15

Don't worry, but I've tried with those values, and it keeps failing, now the right side prints fine, but the left side prints very close to the bed, which means it doesn't print well

abge571, 2022-12-06:
> @abge571 尝试L-0.3 M-0.3
> 
> 抱歉，搞混了。尝试增加L&M，L-0.15 M-0.15

Print the test but now it only prints the upper part well, and the lower left and right part prints very close to the bed, would I need to do another leveling of the bed?

NARUTOfzr, 2022-12-06:
yes. It can be leveled before trying.

  
@abge571 尝试L-0.3 M-0.3
 
抱歉，搞混了。尝试增加L&amp;M，L-0.15 M-0.15

abge571, 2022-12-06:
> yes. It can be leveled before trying.

Okay so I put L-0.3 M-0.3 but level the bed first?

NARUTOfzr, 2022-12-06:
Before leveling, the LM value will be cleared automatically. It can be leveled directly.

abge571, 2022-12-06:
> Before leveling, the LM value will be cleared automatically. It can be leveled directly.

Okay I just tried leveling but it still doesn't work correctly

NARUTOfzr, 2022-12-06:
软件补偿的效果会有偏差。很抱歉，这个方法没能帮到您。

[image]
根据EF的数值：尝试这个方法 #20 。

根据AC的数值，应该尝试拧松A处热床底下的螺丝，将柱子垫高0.3mm。C处的螺丝尽量拧紧。

根据BD的数值，应该尝试拧松B处热床底下的螺丝，将柱子垫高0.5mm。D处的螺丝尽量拧紧。

NARUTOfzr, 2022-12-06:
#10 还有根据这个方法，将黄色部分的V轮放松。
[image]

以上方法为物理方法。不需要设置“M853 L M”

abge571, 2022-12-06:
> 软件补偿的效果会有偏差。很抱歉，这个方法没能帮到您。
> 
> [image] 根据EF的数值：尝试这个方法 #20 。
> 
> 根据AC的数值，应该尝试拧松A处热床底下的螺丝，将柱子垫高0.3mm。C处的螺丝尽量拧紧。
> 
> 根据BD的数值，应该尝试拧松B处热床底下的螺丝，将柱子垫高0.5mm。D处的螺丝尽量拧紧。

Ok I'll try that, but now I tried to level the bed and test the file you gave me with M853 L-0.15 M-0.15 and it works correctly, but when I try it in a gcode file to print something for me it doesn't print well

abge571, 2022-12-06:
> #10 还有根据这个方法，将黄色部分的V轮放松。 [image]
> 
> 以上方法为物理方法。不需要设置“M853 L M”

Okay i'll try that

abge571, 2022-12-06:
> Ok I'll try that, but now I tried to level the bed and test the file you gave me with M853 L-0.15 M-0.15 and it works correctly, but when I try it in a gcode file to print something for me it [...]

Nothing has worked for me, anyway, tomorrow I will try everything you told me, if not, would there be any problem in asking you again?

NARUTOfzr, 2022-12-06:
> Nothing has worked for me, anyway, tomorrow I will try everything you told me, if not, would there be any problem in asking you again?

Of course. I hope I can help you with your problem.

abge571, 2022-12-06:
Give me a moment and I'll send you a photo

>

abge571, 2022-12-06:
> > > 
> > 
> > 
> > I calibrated the arms of the printer and the bed as you told me, but the only file that prints well is the one you gave me... But then when I print something, it doesn't print well... But the leveling is still wrong
> 
> #20 If you adjust it based on this video. The leveling values of EF will be very similar. Could you take a picture to see your leveling value? And the actual state of the print model?

I have been looking at the arm and it is true that even adjusting it it continues to print badly and the values ​​are still similar to those of yesterday
[image]

NARUTOfzr, 2022-12-06:
Good morning! This value is better than the previous one, indicating that the adjustment has some effects. However, there is still some deviation, which still needs to be adjusted.

abge571, 2022-12-07:
> 

Good morning, now what should I do? Should I keep playing the eccentrics?

abge571, 2022-12-09:
> Good morning! This value is better than the previous one, indicating that the adjustment has some effects. However, there is still some deviation, which still needs to be adjusted.
> […](#)

good afternoon, I've been trying to adjust it since the other day, but there's no way, I don't know what to do if a photo is necessary, I'll send it

NARUTOfzr, 2022-12-09:
yes. If possible, please take some pictures. Maybe you can try printing some small models.

abge571, 2022-12-09:
> yes. If possible, please take some pictures. Maybe you can try printing some small models.
> […](#)

At the end of so much adjusting and looking at the printer, the files that you sent me to help me have been of great help, apparently now it prints quite well, but the only thing I have left is one last problem, which I don't know if it has been solved, and it prints well on the left and right side but I still don't know the top and bottom side, I will inform you, thank you very much

cristian-ivascu, 2022-12-13:
What axes do M & L control? 
I've updated to 1.0.5 Beta 6x6 FW in an attempt to dial in settings further. I've made sure the bed spacers are the same length and adjusted all eccentric nuts as well.
Still, my bed seems warped in 1 corner & overall tilted. Since it is the 6x6, I pulled the ABL values using OctoPrint terminal and added the screenshot below.

I'd like to know what M & L control in which axis. The goal would be to compensate towards the back left corner

[image]

NARUTOfzr, 2022-12-13:
As shown in the figure, assuming that the X axis has a tilt similar to that shown in the figure, if the G29 (or M420 S0) is not used for automatic leveling, and the M853 L - (0~1) code is used, the Z axis will perform tilt compensation to make the nozzle always parallel to the hot bed.

 
What axes do M &amp; L control?

 Still, my bed seems warped in 1 corner &amp; overall tilted. Since it is the 6x6, I pulled the ABL values using OctoPrint terminal and added the screenshot below.
 
I'd like to know what M &amp; L control in which axis. The goal would be to compensate towards the back left corner

NARUTOfzr, 2022-12-13:
I have done a lot of tests on this scheme, which will have some effect, but I am not sure whether it can help you.
You can try to set<M853 L-0.15&gt;before G28.

smilingjackal, 2023-01-17:
Wondering if I have this straight. I can add M852 code in Gcode (with the number being desired layer height) after the load current mesh command and the z-wobble type effect will stop. Is that correct?