NARUTOfzr/Neptune_3 #43: Enable filament runout detection capability for OctoPrint

Signum, 2022-11-20:
Firmware: 1.0.5beta

Currently the filament runout sensor only works for prints from SD card. Octoprint cannot detect runouts.
I don't know much about GCODE but it seems that the M115 command gets all the printer's capabilities. One of them is:

    Cap:RUNOUT:0

And that seems to mean that no runout detection is available over USB printing (OctoPrint).

The Neptune 3 has such a nice and sophisticated filament sensor. For me it's the last puzzle piece to be able to leverage it in OctoPrint. Do you think that could be implemented?

Manpoozle, 2022-11-20:
Before you get too invested. The filament sensor has been responsible for about 1 in 15 or 1 in 20 unrecoverable failures in my prints. On the other side of the fence, every time it catches an actual issue I have not been able to recover (4 or 5 times, SD and USB). I now use Michael's firmware fork that turns it off. I felt as you did, and now I see it as a liability. I'd recommend turning it off until such a time that Elegoo engineers can fix it.

cuesir, 2022-12-28:
i've compiled my own firmware for elegoo neptune 2s and the filament runout works great by default marlin. I'm a little lazy to tinker with the neptune 3. But if Elegoo cares enough, just throw in the default marlin codes for filament runout and it should work fine.

mlee12382, 2022-12-28:
> i've compiled my own firmware for elegoo neptune 2s and the filament runout works great by default marlin. I'm a little lazy to tinker with the neptune 3. But if Elegoo cares enough, just throw in the default marlin codes for filament runout and it should work fine.

Unfortunately the N3 runout sensor is handled by the screen firmware so enabling it in the board firmware causes issues.

NARUTOfzr, 2022-12-31:
Yes, as Michael said!

gbarbehenn, 2024-03-10:
So, what does the Settings-Filament Detector setting do, on an Elegoo Neptune 3 Pro?