NARUTOfzr/Neptune_3 #53: M600 color change on Neptune 3 (standard)

TheFeralEngineer, 2022-12-12:
I'm trying to set up M600 for color changes between layers, running programs from Octoprint. The M600 command will work to position the tool head and unload the filament, but nothing will resume. I tried using M0 and it will pause and resume with that, but it doesn't seem to finish the print correctly. 

I wish the firmware on the Neptune 3 Pro were the same as the one on the Neptune 3. It would be nice to be able to run marlin 2.1.1, like the Neptune 3 Pro is running, instead of 2.0.9.3.

TheFeralEngineer, 2022-12-12:
Mine goes through the routine, it's just that it locks up and won't resume when I hit continue/resume on both octoprint and the dwin screen. M0 worked but it seemed like it didn't extrude after it continued. I need to pursue this a little deeper or I'm totally switching to klipper 😆

yakoder, 2022-12-12:
Just curious, what FW version are you running now?

NARUTOfzr, 2022-12-12:
I'm curious why the operation of resuming printing after pausing is unable to extrude. What firmware version do you use? In the latest beta version, I tried to add M0 to gcode to pause. Click the screen to resume printing.

David-McKenna, 2022-12-14:
I ran into the issues described on earlier versions of 1.0.5 and 1.0.3 and had to restart the printer if I tried to pause it, but the latest version of 1.0.5 has M0 working smoothly in my experience. However the load/unload buttons seemed to introduce an additional Z height offset that wasn't removed, so I stuck with manually opening the extruder gear to get things to work.

Ignoring any changes that need to be made to accomodate the screen (if I remember right a few user confirmation screens in the default DGUS firmware are disabled[1][2], though there is still a definition for a filament change page that might work[3]?), the comment needs to be removed on this line[0] in order for Marlin to try handle and M600 calls. `git blame` indicates there were some tweaks made to the M600 parameters in the more recent 1.0.5 work, but M600 was never enabled.

[0] https://github.com/NARUTOfzr/Neptune_3/blob/main/Source%20Code/ZNP_F401_Marlin-Beta/Marlin/Configuration_adv.h#L2517

[1] https://github.com/NARUTOfzr/Neptune_3/blob/main/Source%20Code/ZNP_F401_Marlin-Beta/Marlin/src/lcd/extui/dgus/DGUSScreenHandler.cpp#L85
[2] https://github.com/NARUTOfzr/Neptune_3/blob/main/Source%20Code/ZNP_F401_Marlin-Beta/Marlin/src/lcd/extui/dgus/dgus_extui.cpp#L61

[3] https://github.com/NARUTOfzr/Neptune_3/blob/main/Source%20Code/ZNP_F401_Marlin-Beta/Marlin/src/lcd/extui/dgus/mks/DGUSDisplayDef.h#L200

TheFeralEngineer, 2022-12-15:
> 

I'm using 1.0.5 beta, but I modified it a little (I have the forked version in my repository). I only tried to use M0 once, I was more curious about getting M600 to work properly from Octoprint. It will not resume from Octoprint or from the Dwin screen.

TheFeralEngineer, 2022-12-15:
> I ran into the issues described on earlier versions of 1.0.5 and 1.0.3 and had to restart the printer if I tried to pause it, but the latest version of 1.0.5 has M0 working smoothly in my [...]

Thanks! I'll take a look into that this weekend.

FoxxMD, 2023-05-22:
Has there been any progress on getting m600 working?

Ok2dot0, 2024-11-30:
Are there any updates?

TheFeralEngineer, 2024-11-30:
This machine has been discontinued for over a year. The only way to get
updates will be from community firmware or switching to klipper
Phil T.
The Feral Engineer

www.theferalengineer.com

Catnip contributions 🐱🐱
https://paypal.me/timpsonmusicllc