NARUTOfzr/Neptune_3 #55: Recovery from runout or pause function fails

portudork, 2023-01-03:
Whenever I pause a print to change filament or trip the filament runout sensor the printer fails to resume printing (extruding filament) upon resume. The print head goes back to the last spot before pause but doesn't extrude any filament anymore while the print head moves effectively ruining any print. The pause and runout function are effectively useless due to this issue.

NARUTOfzr, 2023-01-03:
Try making the change:
[image]

[image]

Or try updating firmware V1_1.0.5-beta

portudork, 2023-01-04:
Do you mean to uncheck that setting and remove that g code? By default it's checked and that code is already in there.

NARUTOfzr, 2023-01-04:
Yes. This is not the configuration file uploaded by elegoo. It does not match the machine. Extrusion relative mode should not be used.

portudork, 2023-01-04:
And delete out the G92 E0 as well right? Over you confirm I'll do both and retest as soon as this print is done. Thanks for the quick response!

NARUTOfzr, 2023-01-04:
yes. You can also comment out this line by adding ";" before it.

TheAussieTinker, 2024-05-23:
I have done what you said, unchecked use relative e distance and commented out all G92 E0 lines in the gcode on PrusaSlicer. However when I open the gcode file PrusaSlicer has injected heaps of G92 E0 throughout the file that aren't commented out.

[image]
[image]