NARUTOfzr/Neptune_3 #64: Firmware update instructions should be bytes not bits (at least on mac)

chrishoina, 2023-02-02:
I successfully updated the firmware, but needed to use 4096 bytes, not bits for the allocation size. Attached are my ending settings. Device block size remains 512 bytes, however allocation size was changed to 4096 bytes. 

[image]

NARUTOfzr, 2023-02-21:
Sorry for the confusion. It is 4096 bytes.

chrishoina, 2023-02-22:
Awesome, thanks for the confirmation. I hope this helps, the existing documentation on the Eleggo site was a little confusion (especially for somebody doing this the first time). Thanks again!