NARUTOfzr/Neptune_3 #69: Z layers not printing properly

TheAussieTinker, 2023-05-05:
On the left is printed on my prusa mk3 and looks as it should on the right is printed on my elegoo Neptune 3. It seems like the z axis has squished many layers into one of the middle layers before continuing, resulting in this weird seam and a piece that has incorrect dimensions.
My setup:
- Elegoo Neptune 3
- latest board firmware (V1_1.0.5-beta)
- raspberry pi running octoprint running the prints
- printing with ESUN PLA+

What I have tried:
- re-slicing many times
- using cura instead of prusa slicer and re-slicing
- tightened the z collar
- tightening and loosening belts
- changed filament (always printed with pla)
- checked the filament can be pulled easily

How should I fix this?

[image]

NARUTOfzr, 2023-05-06:
hi @TheAussieTinker 
在调平界面检查网床数值，排除是否网床数据偏差太大导致？（通常我们控制网床左右偏差±0.5以内最好）如果您遇到网床倾斜的问题，可以使用以下方法校正：https://youtu.be/MRsDb_wC8S8

NARUTOfzr, 2023-05-06:
hi @TheAussieTinker
1. Check the grid value in the leveling interface to rule out whether the grid data deviation is too large? (Usually we control the deviation between the left and right values of the grid within ±0.5 is best.) If you encounter the problem of hot bed tilt, you can use the following method to debug:   https://youtu.be/MRsDb_wC8S8

2. Manually move the Z-axis up and down to rule out whether the stepper motor loses steps because of moving resistance? If so, the Z-pulley needs to be properly adjusted according to the following method:
[Neptune3-Leveling Issue Solutions.pdf](https://github.com/NARUTOfzr/Neptune_3/files/11410826/Neptune3-Leveling.Issue.Solutions.pdf)

3. You can also increase the Z-axis drive current appropriately, adjust the position of the potentiometer as shown in the figure, decrease clockwise, increase counterclockwise, and fine-tune appropriately (pay attention to turn off the power operation).
[image]

TheAussieTinker, 2023-05-06:
I think it may be an issue with the V1_1.05-beta firmware working with octoprint. I used a version of the firmware modified by Michael Lee and The Feral Engineer. I have positive results so far but I am going to do some more tests before I post about it.

TheAussieTinker, 2023-05-07:
FIXED: This was an issue with the firmware interfacing with octoprint. In order to resolve this issue I built a version of marlin from The Feral Engineer (link below) and commented out the mods he added. I then flashed the firmware onto the printer, releveled the bed, and it was good to go. Working properly with octoprint

I used the source code from the ZNP_F401_Marlin-Beta folder
https://github.com/TheFeralEngineer/Neptune_3

closed 2023-05-07 by TheAussieTinker

NARUTOfzr, 2023-05-08:
@TheAussieTinker Thank you very much for your feedback. This helps us a lot and we will check and fix it.  :heart: