NARUTOfzr/Neptune_3 #71: Neptune 3 Issues with PETG prints

RichBr1, 2023-07-04:
Cannot get any decent prints out of the Neptune 3 using Cura latest and Neptune 3 profiles. Stringing everywhere and terrible top layers. Have changed everyting in Cura (temp, speed etc) but nothing works,? PLA works a treat.

Moody66, 2023-07-04:
This is not a firmware issue. Neptune 3 prints it fine. Go to the Factbook
page and as about peoples settings.

RichBr1, 2023-07-04:
PLA works perfectly so I Don't think it is firmware, but I have the Elegoo Neptune 3 profiles loaded and have tried Generic PETG ans well as Eryone PETG profiles and have spent hours chaning settings but cannoy get a decnet print. I have some suspicions that it is overextruding as it seems to double feed when it feeds but there are only so many settings you can change. Everyting comes out very stringy and walls separate in places, just need to find the magic setting in Cura or other slicer. Maybe I can post in discussions.

adam120800, 2023-07-13:
PETG is tricky, even though they are posts about it being hard. Its harder than uncolored ASA to print usually. And don't even try getting perfect prints with transparent PETG. It doesn't like to be squished even the slightest, so your Z-offset needs to be different than PLA, this is the issue most people don't mention. Use a flat rectangle model with height 0,24 or whatever your layer height is, and then adjust the nozzle on touchscreen while its printing until you can see clear separation of lines on first layer or brim then lower it until the lines meld together ever so slightly, also your finger once it passes there will be ridges but minor (use 0.01mm increments please). Don't mistake a mirror finish on first layer as a good result for PETG due to line squish. 
   Prusa printers which are regarded as properly calibrated unless you have bad luck, are set to have as high as possible offset while still printing within spec.  Your retraction is probably not tuned, you can always turn up the retraction speed first to see if it makes a good change. Bowden is limited due to needing massive amounts of retraction for PETG, better keep it at default with +-30% changes if needed (upgrade to direct drive if you have $, you will not regret it for a moment). No Z-hop, use it when you are printing simple geometry without it and without issues. Cooling; PETG can be cooled at 30-40% at first layers to reduce stringing and blobbing not 0% at start (check for strength if you start cooling at first layer!) . You can change top layer thickness and most importantly support interface thickness if the layer has supports under also don't go overboard with 100% cooling on other layers since PETG doesn't require and prints worse with too high of volume.
  Extrusion: from reading most reddit posts it seems that these printers do over extrude by 1-2% and its probably different unit to unit so try to reduce the flow by 2 percent until you reach 96, but only after you verify first layer printing. If you think you need to go under 96 then you have previously mentioned issues. Most Filaments need different flow rates!
  Speed: You need to have travel moves above 150 at least to get reduced blobbing and ableist 200 to get reduced stringing. You can print faster once you prints start to clear up and get better. Don't get bogged down by ender 3 users telling you have to go under 70mm/s to get good and even detailed prints. You can print faster until you vibrate the detail out of the print pretty much. I hope we will soon see 2.1.2 for input shaping to mitigate this.
  Heat: my culprit for printing bad PETG, most PETG unless it is more G than PET needs to be printed at 230-235C+ (i print at 241 due to steel nozzle), if you go under 230 you usually lose strength, bad first layers but with reduced blobbing stringing. Linear advance: I have mine at 0.08 but I feel 0.07 is more safe and probably good for 99% of ppl with these printers. Warning: if you haven't used it in the past, it makes you print head sound like a small fax machine, don't be alarmed. 

Once I got all this down I felt like I could print with nails instead of filament with 160mm/s speed. My N3Max bed cant handle more that 170mm/s on long travel moves but smaller beds can go much faster, for speed and travel.

Since I was in the mood and I was keen to write a novel because i remember my recent struggles but probably wont repost it or comment anywhere else so feel free to copy, paste modify to help others. I am frustrated that there are a lot of youtubers that give people impressions that PETG is easy to print while they video their models from 5m and more. I recommend this Chad for calibration (https://www.youtube.com/watch?v=rp3r921DBGI) or NARUTOfzr when he has free time to assist and this video (https://www.youtube.com/watch?v=vSwumoSlZTo) to have an idea of how prints should look like with different materials; he has a more expensive and properly calibrated printer than you and me will probably ever need and he honestly presents his prints.  Sidenote: All of this goes in the water if your printer is not mechanically sound, tightened bed v-slot wheels and screw checked stuff that should be tight (Z-axis bushing screws should be semi-lose), lose belts, lemon printer, bent gantries or leadscrews in the worst scenario which I don't think you have at all :).