NARUTOfzr/Neptune_3 #74: Not possible to reduce flow during print via display

h3nk86, 2023-07-28:
Hi. I´m running FW 1.1.5.3 with UI Version 1.5beta02. It is possible to increase flow during print, but not to decrease it (what is more important to me). I updated the UI Version because of that problem with no luck.
Video of the issue:  [https://youtube.com/shorts/xo_je5g31dA?feature=share](url)

Hope for support.

Moody66, 2023-07-29:
You can change the flow. But you can also just change it in your slicer. OR calibrate your flow for your printer.

h3nk86, 2023-07-30:
Thank you moody66.  Of course I know all of that.  it is explicitly about influencing the flow DURING the printing process. I believe it worked fine with another FW.
I think the link for the video is not working properly:  [Klick](https://youtube.com/shorts/xo_je5g31dA?feature=share)

NARUTOfzr, 2023-07-30:
hi! @h3nk86 This is an error on the Neptune3Pro/plus/max firmware, it is recommended to try this firmware:  https://github.com/NARUTOfzr/Neptune_3_Pro_Plus_Max/releases

h3nk86, 2023-07-30:
Thank you. 
I´m looking forward for the fix in future updates
 Henk

closed 2023-07-30 by h3nk86